leggedrobotics/wild_visual_navigation #304: Collision still exists after 8 minutes of online training

jmwang0117, 2024-04-19:
[image]
[image]

Thank you for your great work!

In your demo, I control Jackal to move for 8 minutes. The prediction results can indeed show the passable area and the impassable area, but the robot's behavior is not normal. Jackal will still hit the tree without taking any measures to avoid obstacles. Measures, how to solve this problem?

mmattamala, 2024-04-19:
Hi @jmwang0117, thanks for your interest in WVN, we are glad to see that the demo worked for you.

The behavior you observed is expected, as this demo's only goal is to showcase the online traversability training. The simple interface we implemented here to control the robot is not exploiting the traversability prediction, so it will not avoid obstacles.

To achieve the closed-loop navigation behavior, the prediction needs to be integrated with a navigation pipeline. For our experiments we use a local elevation mapping module + a local planner. 

We did not add them here on purpose because we tried to keep the demo minimal, and also because these components might change depending on the robot setup (e.g. some people do local planning on image space, or they might to use their own local planner).

mmattamala, 2024-04-25:
Sorry for the delay, a few answers:

> Is there a complete navigation demo in this code base? Or is there any other work that implements navigation based on "Traversability Estimation"?

No, there is no local planning implemented in this repository. We might add an example building top of the [elevation_mapping_cupy](https://github.com/leggedrobotics/elevation_mapping_cupy) package as we did for this work but we don't have a timeline for this at the moment.

> (b) Some of the SDFs generated from the predicted traversability during autonomous operation.

This is part of the same local planning pipeline I referred above.

> (c) Global 2.5D reconstruction of the testing area and predicted traversability, generated in post-processing to illustrate the capabilities of our approach

This is not a feature of the method, but simply post-processing we did to illustrate the resulting traversability map in a better manner. We did not use this for navigation. You can generate a similar result if you configure the aforementioned `elevation_mapping_cupy` package with a large grid, but we do not recommend this for real navigation tasks.

closed 2024-12-20 by jmwang0117