leggedrobotics/wild_visual_navigation #305: About elevation mapping

lyysl, 2024-06-07:
I would like to ask is the elevation mapping available?
I saw /elevation_mapping/semantic_map in the rostopic list but there is nothing output in 
rostopic echo /elevation_mapping/semantic_map. And there is the option in Rviz but does not display anything for elevation map. Thank you

mmattamala, 2024-06-07:
Hi @lyysl, I assume you are trying the simulation example in the container?

We can add it to the container in the next few days, but we used [elevation_mapping_cupy](https://github.com/leggedrobotics/elevation_mapping_cupy) if you are interested.

lyysl, 2024-06-07:
I followed the ROS setup and using the ROS bag you provided in google drive for real scenes.

lyysl, 2024-06-07:
Or it needs point cloud data which is not in the bags?

mmattamala, 2024-06-16:
Hi @lyysl Apologies for the delay

The point cloud data does exist in the bags. For example, for the Banhofstrasse bag (`2023-08-19-16-45-16_Bahnhofstrasse_Zurich.bag`):

```sh
/depth_camera_front_lower/point_cloud_self_filtered    2023 msgs    : sensor_msgs/PointCloud2                
/depth_camera_front_upper/point_cloud_self_filtered    2023 msgs    : sensor_msgs/PointCloud2                
/depth_camera_left/point_cloud_self_filtered           2023 msgs    : sensor_msgs/PointCloud2                
/depth_camera_rear_lower/point_cloud_self_filtered     2023 msgs    : sensor_msgs/PointCloud2                
/depth_camera_rear_upper/point_cloud_self_filtered     2023 msgs    : sensor_msgs/PointCloud2                
/depth_camera_right/point_cloud_self_filtered          2024 msgs    : sensor_msgs/PointCloud2                
/motion_reference/command_twist                        1348 msgs    : geometry_msgs/TwistStamped             
/point_cloud_filter/lidar/point_cloud_filtered         1348 msgs    : sensor_msgs/PointCloud2  
```

To run the elevation mapping cupy with these data, you can run this launchfile: [elevation_mapping_cupy.launch](https://github.com/leggedrobotics/wild_visual_navigation/blob/main/wild_visual_navigation_anymal/launch/elevation_mapping_cupy.launch)

lyysl, 2024-07-09:
I installed eleveation_map_cupy and build. And I run
`roslaunch wild_visual_navigation_anymal elevation_mapping_cupy.launch`.
However, I got an error
```
[ INFO] [1720522902.646004305]: Subscribed to PointCloud2 topic: /point_cloud_filter/lidar/point_cloud_filtered
[elevation_mapping-1] process has died [pid 6974, exit code -6, cmd /mnt/c/myProject/phd/what/software/terrain_segmentation/wild_visual_navigation/catkin_ws/devel/lib/elevation_mapping_cupy/elevation_mapping_node __name:=elevation_mapping __log:=/home/simon/.ros/log/54bcccb4-3ddf-11ef-9fb2-00155d257914/elevation_mapping-1.log].
log file: /home/simon/.ros/log/54bcccb4-3ddf-11ef-9fb2-00155d257914/elevation_mapping-1*.log
all processes on machine have died, roslaunch will exit
```
Thank you!

JonasFrey96, 2024-07-20:
Hi, 
were you able to run the master branch of elevation_mapping_cupy standalone? 
If not, you can move the issue to the elevation_mapping_cupy repo?

closed 2024-08-02 by lyysl