leggedrobotics/wild_visual_navigation #309: IndexError: list index out of range

Yinmlmaoliang, 2024-07-11:
When I start the Gazebo simulation environment and the RViz window as the main interface.
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.20.24.207
[Msg] Loading world file [/root/catkin_ws/src/wild_visual_navigation/wild_visual_navigation_jackal/worlds/outdoor.world]
[ INFO] [1720660848.245358585]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1720660848.364386093]: Physics dynamic reconfigure ready.
[ERROR] [1720660848.613223, 0.001000]: bad callback: <function gazebo_callback at 0x7fe93ed40e50>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/root/catkin_ws/src/wild_visual_navigation/wild_visual_navigation_jackal/scripts/carrot_follower.py", line 55, in gazebo_callback
    robot_x, robot_y, robot_yaw = msg_to_se2(msg.pose[1])
IndexError: list index out of range

[ERROR] [1720660848.613761, 0.001000]: bad callback: <function gazebo_callback at 0x7f8cfd6b91f0>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/root/catkin_ws/src/wild_visual_navigation/wild_visual_navigation_jackal/scripts/gazebo_world_publisher.py", line 65, in gazebo_callback
    T_world_base = msg_to_se3(msg.pose[1])  # this is the base_link pose in world frame (from gazebo)
IndexError: list index out of range

closed 2024-07-11 by Yinmlmaoliang

nankaixiaokai, 2024-07-19:
How did you solve it？

mmattamala, 2024-07-19:
It's very likely that something didn't launch correctly - because the carrot follower is subscribing an apparently ill-defined message.