leggedrobotics/wild_visual_navigation #310: Traversability value saturation

Matthewjsiv, 2024-08-19:
Thanks for your work on open sourcing this and making it easy to run! I just had a quick clarification question regarding the traversability supervision signal.

I noticed on both of the outdoor bags that the value published to "/wild_visual_navigation_node/instant_traversability" are generally saturated in the range of .98-1.0. I am wondering if this is expected or I have something set up wrong on my end, as in the paper it is mentioned that the values are calibrated based on the platform to stretch between 0-1.0.

Additionally, I was wondering if you guys are planning on releasing anymore bags with more diverse robot-environment interactions, for example the experiments in Figs 1&11 from the 2024 paper.

Thanks again!

mmattamala, 2024-09-06:
Hi @Matthewjsiv, apologies for the very late reply.
If I remember correctly, the sequences in the rosbags are not hard enough to be make that distinguishable. If you want to make it more sensitive, you cna change the parameters of the sigmoid function, which unfortunately we did not expose in the config files: https://github.com/leggedrobotics/wild_visual_navigation/blob/8b9caf915e668653b761b57ed33d2719a38b4bfd/wild_visual_navigation_ros/scripts/wvn_learning_node.py#L95

Regarding the rosbags, we are not planning to publicly release more bags at the moment. There are other ongoing projects (especially on @JonasFrey96's side) which involve more interesting data to be released but we don't have a timeline for this I'm afraid.

JonasFrey96, 2024-12-13:
@Matthewjsiv we are currently having a big push to open-source a dataset with over 100 environments. The data should be officially ready by Feb. 2025. I hope this will be still helpful for you at this point.