leggedrobotics/wild_visual_navigation #312: opencv-python version mismatch between wild-visual-navigation and stego

AfonsoEloy, 2024-11-27:
Hi guys! First of all, thank you for the great work.

I am following the docker setup instructions on a 20.04 Noetic machine. Everything appears to work fine during the build and first setup, except this error that pops up:

`ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
wild-visual-navigation 0.0.1 requires opencv-python==4.2.0.34, but you have opencv-python 4.6.0.66 which is incompatible.`

If, instead, I try to force the installation of 4.2.0.34, then I receive a similar error complaining that stego requires >= 4.6.

Am I doing something wrong or is there something wrong in the setup files?

Thanks in advance!

mmattamala, 2024-11-27:
Hi Afonso, thanks for your interest in our paper!

We're a bit busy at the moment, but I can take a look early next week if that works. I agree that there could be a dependency mismatch that we oversaw. Pinning the versions might solve it but I'll need to test it to see if we need to propagate corrections to other repos.

I hope that works, but let me know if you are in a rush

mmattamala, 2024-11-29:
Hi @AfonsoEloy , I just pushed a minor fix in the `self_supervised_segmentation` repo to relax the opencv version, which was set to >=4.6. I tested it locally on my laptop and it works as intended, let me know if the problem persists.

AfonsoEloy, 2024-11-29:
Hi @mmattamala. Your fix solved the issue, thanks! Final set of questions:

1. Is the docker container as-is suitable to run WVN with a dataset of real data, or only for the simulation demo?
2. The datasets we are collecting only have images from a single RGB-D front-facing camera (realsense d435i). Considering I see multiple front and rear cameras being referenced in your config files, can the system work with only one?
3. Aside from the images, are there any mandatory inputs that must come from a specific platform like ANYmal, or do other legged robots work (such as the Unitree Go1 EDU, which we are using).

mmattamala, 2024-11-29:
Awesome, glad to know it worked. Regarding your questions:

1. The docker container should work with real data too, unless there are specific packages and dependencies required by your data (e.g., message definitions). I should have added a devcontainer config file to make the development easier (so you work with VSCode within the container, but I learned this after I prepared this docker). I could do it next week if you think it might help you
2. That's absolutely possible and it's actually the configuration we used for our RSS paper. The multi-camera feature was added in our extended paper but it still processes one camera at a time
3. If you inspect the Jackal simulation example more closely, it only requires the image, tf data with an odometry source (to generate the supervision signal via the reprojected path), and a velocity command. We did not use any raw proprioceptive signal (e.g., IMU or joint encoders). Jonas has a follow up paper using proprioception to estimate physical terrain properties: https://leggedrobotics.github.io/identifying_terrain_physical_parameters_webpage/

AfonsoEloy, 2024-12-02:
Don't worry about the devcontainer, I can manage without it. I'll spend the next few days working on integrating your system with our data. Your help has been incredibly valuable -- thank you!

mmattamala, 2024-12-03:
Awesome, you're welcome! I look forward to seeing how WVN works with your system :)
I'll close the issue for now but feel free to reopen it if new problems arise.

closed 2024-12-03 by mmattamala