leggedrobotics/wild_visual_navigation #316: Branch for cupy Integration Missing mask_elevation_semantics Plugin

LShivaRudra, 2025-03-03:
Hi everyone,

I was looking into the integration with cupy and couldn't find the mask_elevation_semantics plugin in the main branch of cupy. Could you clarify which branch was used for this integration?

Also, is this something that you customized and haven't open-sourced yet?

Thanks!

JonasFrey96, 2025-06-05:
I am sorry for the late response. 

We ran the following branch: https://github.com/leggedrobotics/elevation_mapping_cupy/blob/dev/set_default_to_nan/elevation_mapping_cupy/script/elevation_mapping_cupy/plugins/mask_elevation_semantics.py

closed 2025-06-05 by JonasFrey96

fwx20020105, 2026-05-25:
> Hi everyone,  大家好，
> 
> I was looking into the integration with cupy and couldn't find the mask_elevation_semantics plugin in the main branch of cupy. Could you clarify which branch was used for this integration?我在研究与 Cup 的集成时，找不到 Cup 主分支里的 mask_elevation_semantics 插件。能否说明这次集成使用的是哪个分支？
> 
> Also, is this something that you customized and haven't open-sourced yet?另外，这是你定制但还没开源的东西吗？
> 
> Thanks!  谢谢！

I'm sorry to bother you.I found that the Google Drive links for checkpoints and rosbags are returning 404 errors. I was wondering if you still have the relevant field rosbag files available. If so, could you please send them to my email address: 1035664828@qq.com? I would be extremely grateful.